BossHobby/QUICKSILVER #33: feature rc channel order

andreyk72, 2021-11-11:
Hi, Please regard possibility to implement RC axis channel order mapping switch (AETR / TAER). 

This can be convenient for pilots who use one model for all their fleet and eventually use TAER channel order instead of AETR that QS supports by default.

bkleiner, 2021-11-18:
from discord discussion:
in the past the argument was made that this should be handled by the TX and QS should just pick one and stick to it.
if more users feel this should be handled in the FW we will consider it for implementation.

Sergii-Goncharov, 2021-11-26:
> from discord discussion: in the past the argument was made that this should be handled by the TX and QS should just pick one and stick to it. if more users feel this should be handled in the FW we will consider it for implementation.

I think this should be handled by QS. In my TX I use one model for all kwads

bkleiner, 2022-03-22:
turns out there are transmitters which do not allow you to easily swap the channel order. the emax e8 seems to be such a case.

bkleiner, 2022-06-19:
hit develop in https://github.com/BossHobby/QUICKSILVER/commit/cc5696980cbc788652ac2594f8ff5586d7f1ca58 and following, should ship in v0.8.5-dev

closed 2022-06-19 by bkleiner