BossHobby/QUICKSILVER #37: Tramp broken on SUPRA VTX

bkleiner, 2021-11-18:
Device seemingly responds, but does not come back with data on the Tramp requests
```
UNIFIED: trying protocol 3
>> TRAMP: timeout waiting for packet
>> UNIFIED: trying protocol 4
>> TRAMP: timeout waiting for packet
>> TRAMP: timeout waiting for packet
>> UNIFIED: trying protocol 5
>> TRAMP: timeout waiting for packet
>> TRAMP: timeout waiting for packet
>> SMART_AUDIO: send cmd 0 (7)
>> SMART_AUDIO: mirror 0x75 (0)
>> SMART_AUDIO: invalid mirror (0:0x75)
>> SMART_AUDIO: send cmd 0 (7)
>> SMART_AUDIO: mirror 0x0 (0)
>> SMART_AUDIO: mirror 0xaa (1)
>> SMART_AUDIO: mirror 0x55 (2)
>> SMART_AUDIO: mirror 0x3 (3)
>> SMART_AUDIO: mirror 0x0 (4)
>> SMART_AUDIO: mirror 0x9f (5)
>> SMART_AUDIO: mirror 0x0 (6)
>> SMART_AUDIO: magic 0x0 (0)
>> UNIFIED: trying protocol 6
>> SMART_AUDIO: timeout waiting for packet
>> SMART_AUDIO: send cmd 0 (7)
>> SMART_AUDIO: mirror 0xaa (0)
>> SMART_AUDIO: mirror 0x55 (2)
>> SMART_AUDIO: mirror 0x3 (3)
>> SMART_AUDIO: mirror 0x0 (4)
>> SMART_AUDIO: mirror 0x9f (5)
>> SMART_AUDIO: mirror 0x0 (6)
>> SMART_AUDIO: magic 0x0 (0)
>> SMART_AUDIO: timeout waiting for packet
```

Noramas, 2021-12-21:
👍

bkleiner, 2022-01-10:
fixed in thanks to @Noramas for many many flashes :D
https://github.com/BossHobby/QUICKSILVER/commit/0efd33228c8981515fc0ac9a7f3c71a9fe9abf6a

closed 2022-01-10 by bkleiner