BossHobby/QUICKSILVER #65: Turtle Mode sometimes does not reverse motor direction

Stampede10343, 2022-08-31:
I have ran into this issues on both Crazybee FR and ELRS targets, with a switch and with the auto ON detection.

Many times the quad will arm, motors will freak out, then it will say turtle, motors will pulse and alternate, then stick inputs cause the correct motors to spin the wrong direction. I can feel the air as well as see the appropriate side of the quad get pushed into the ground.

When this happens occasionally a disarm then rearm will fix it, but often I need to pick up the quad.

bkleiner, 2022-08-31:
there were some issues around motor reversal in the past and a  [increased delay was added](https://github.com/BossHobby/QUICKSILVER/blob/master/src/drivers/drv_motor_dshot.c#L358).

i am a little fuzzy on what happens for you in what order. could you please re-iterate on that?
do the motors reverse correctly when entering turtle mode? are you holding the stick in the direction you want to flip?

Stampede10343, 2022-09-01:
Sure, the process is:
1. Crash and land upside down :laughing: 
2. Arm
3. Motors spin after a moment pushing the quad harder against the ground
4. Turtle mode will engage
5. Often it will say turtle and respond to the stick movement, but still push the quad into the ground
6. Sometimes disarming then re-arming fixes it, but often it does not

bkleiner, 2022-10-09:
i have not been able to nail this issue, or even properly reproduce it.

So i have taken another look at the bluejay source code and tried to overhaul the reversal code in a way that _should as far as i can tell_ work. Seeing that this didnt change it in major way from how it worked in the past, i am not sure if it will do anything.
Changes are included in the [latest dev release](https://github.com/BossHobby/QUICKSILVER/releases/tag/v0.9.2-dev)

I am beginning to suspect that maybe is a situation _in bluejay_ where the orientation change gets "blocked" until either a dshot de-sync or a reset occurs. This also makes sense because it seems once QS has failed to change the direction the escs seem to no accept _any_ direction change commands anymore.

There recently were some changes merged to [bluejay](https://github.com/bird-sanctuary/bluejay/pull/20) which all the transmission of additional information via dshot telemetry. i hope i can use this to troubleshoot the problem in further in the future. (gotta implement dshot temetry on my end first)

Stampede10343, 2022-10-09:
Thanks for the update! I'll have to give the new version a try, its pretty finicky to reproduce, but it happens pretty often and requires a disarm and rearm to get it to reverse direction.

bkleiner, 2022-11-26:
in the [latest dev release](https://github.com/BossHobby/QUICKSILVER/releases/tag/v0.9.4-dev), reverting to "regular" direction is re-tried on every arm. The direction change is done in accordance to the specification.
Closing this issue because if it still doesn't work with this change, we probably need to open one with bluejay.

closed 2022-11-26 by bkleiner