BossHobby/QUICKSILVER #73: Battery Low Warning never really shows up

Stampede10343, 2022-12-12:
I've always had issues with the Low battery warning showing up, unless its set way high, e.g. 3.6 volts or so.

Would it be possible to have the battery warning based off the filtered voltage, rather than the 'fuel tank voltage'

bkleiner, 2022-12-13:
i guess the underlying issue is that the fuel gauge value is not very useful as it is right now _in general_.
i will have a dig if there is anything wrong with its implementation.
If there is, let's give the fixed version another chance.
If there is not, i guess we might as well drop it all together.

ChuckBorris33, 2023-11-03:
Would it be possible to add option in configuration to use filtered voltage instead of fuel gauge voltage ? It's a good idea but some of mine batteries don't show warning even if it's set to 3.7V and I loose video before I see warning.  It worked quite well for new tattu batteries.

vad7, 2023-12-14:
The same issue. 
Happymodel SuperX HD 1S (F411).

sdratwa, 2024-02-09:
The same issue. Happymodel Diamond F4. At the beginning I set to 3.3, then 3.5 (default), now 3.7 and even close to 3.0 does not show anything... unless I disarm.

bkleiner, 2024-07-22:
switch landed in https://github.com/BossHobby/QUICKSILVER/releases/tag/v0.11.0-rc2

closed 2024-07-22 by bkleiner