BossHobby/QUICKSILVER #75: GeprcF411 Gyro not found

RPassanha, 2023-01-12:
I have Geprc Smart16. After flash FC , I received the error message below.

Faillop gyro not found (4) Detected!
Please fix the issue to be able to change settings.

My FC is STM32F411 (S411) board_name GEPRCF411.
[image]
[image]

bkleiner, 2023-01-15:
this caused either by the gyro pins being off or this fc using a gyro we dont support.
i am not aware of any gyros being in use we dont support, but please look up in betaflight what gyro is detected there.

as for the pins, sadly these days there is no guarantee anymore that the betaflight target the fc shipped with actually matches the pins in use. i have ran into this before with geprc.
they would start with one of their official targets then apply modifications on top of that. there is no good way for us to support hardware like that.

RPassanha, 2023-01-16:
Betaflight works well. This is Betafligth dump.

# 

# dump all

# version
# Betaflight / STM32F411 (S411) 4.4.0 Jan 11 2023 / 06:08:18 (2417d50d8) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name GEPRCF411
manufacturer_id GEPR
mcu_id 0053004d3535511536323431
signature 

# name: SMART16

# resources
resource BEEPER 1 B02
resource MOTOR 1 B04
resource MOTOR 2 B05
resource MOTOR 3 B06
resource MOTOR 4 B07
resource MOTOR 5 B03
resource MOTOR 6 B10
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 NONE
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
[... 1315 lines cut ...]
set rates_type = ACTUAL
set quickrates_rc_expo = OFF
set roll_rc_rate = 7
set pitch_rc_rate = 7
set yaw_rc_rate = 7
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 67
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998
set roll_level_expo = 0
set pitch_level_expo = 0

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#

bkleiner, 2023-01-16:
the pins seem to check out. what gyro does BF indicate? post the output of "status" on the cli.

RPassanha, 2023-01-17:
# status
MCU F411 Clock=108MHz (PLLP-HSE), Vref=3.29V, Core temp=34degC
Stack size: 2048, Stack address: 0x2001fff0
Configuration: CONFIGURED, size: 3777, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=BMI270, ACC=BMI270
OSD: MSP (30 x 13)
BUILD KEY: 4.4.0-RC4/GEPRCF411/8d08ced0ca99f954123aaf91bbe66297
System Uptime: 22 seconds, Current Time: 2023-01-17T00:18:48.677+00:00
CPU:43%, cycle time: 313, GYRO rate: 3194, RX rate: 15, System rate: 9
Voltage: 146 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
Arming disable flags: RXLOSS CLI MSP RPMFILTER

closed 2023-12-09 by bkleiner