BossHobby/QUICKSILVER #76: ESC firmware corruption during settings write with esc-configurator

regularexpression, 2023-01-15:
1. Flashed Bluejay with the help of  esc-configurator.com on Jhemcu gsf405a aio (Betaflight was installed).
2. Flashed QuickSliver on flight controller.
3. Tried to change motor direction with the help of esc-configurator.com.
4. Three of four esc disappeared.
5. Reflashed Betaflight.
6. Blheli configurator could see all four esc but with corrupted firmware.
[image]
7. Force reflashed to blheli_s with Blheli configurator.
8. All four esc appeared in esc-configurator.com.

This was tried 2 times.

bkleiner, 2023-01-15:
what QS version did you run? 0.9.5?

regularexpression, 2023-01-15:
> 

yes

bkleiner, 2023-01-15:
does not appear to be a general issue, i just ran through a couple of scenarios.
might be related to the JHE board somehow. those don't have the best track-record.

regularexpression, 2023-01-16:
I'll try to investigate this problem more this week.

voyn238, 2023-01-22:
A similar thing happened to me with my Mobula 7 1S ELRS on their X12 board. ESC 2 got wiped out and it took me a bunch of tries via different methods to get BLHeli_S back on it, and then re-flashed to BlueJay. Scary! I thought my board was toast.

bkleiner, 2023-01-31:
Please keep me up-to-date if you were able to gain more insight into what is happening here.
I can't seem to reproduce the issue with boards i have on hand here (crazybee, bfpv elrs, matekf405).

regularexpression, 2023-01-31:
Sorry for the delay. Had less time then expected.
Just tested four different crazybee_f4_elrs(x12, elrsf4 2g4, diamond and superbee), one crazybee_f4 (crazybee f4 pro + external elrs on uart) and two Jhemcu gsf405a(with different gyros).
All AIOs were powered by power supply with fixed voltage 4.0v (not battery).
Problem randomly appears only on jhemcu, but this time I can flash them via esc-configurator.com.
Problem appears only when QuickSilver flashed (v0.9.5, v0.9.5dev, v0.8.5-dev were tested).
With Betaflight v4.4.0 flashed problem doesn't appear (reflash from QS to BF and vice versa was done several times).
According to my observations during "Connect" button click in esc-configurator.com something goes wrong (not correct sound from esc).
[image]

regularexpression, 2023-01-31:
Also want to note that Jhemcu gsf405a and superbee f4 have the same esc firmware S-H-50.

regularexpression, 2023-01-31:
And maybe someone with  GOKU Versatile F405 1-2S 12A (FLYWOOF405S_AIO) could try to repeat the same steps, because this FC is very similar to jhemcu.

banyaszg, 2023-02-06:
I don't know whether there is a connection but in the case of my Beta F4 1S serial ELRS if I load the ESC info in the QS configurator, then it won't work. I receive only corrupt info. But for me the ESC works well.
(QS 0.9.5, ESC A-X-5, BJ 0.18.1)

regularexpression, 2023-02-07:
I've checked Betafpv f4 1S 5a aio (Spi elrs) on QS 0.9.5 (esc o-h-5) - no issue.

bkleiner, 2024-06-27:
i might have come across something previously QS would feed the blheli response bytes as they came to the usb serial port, which, depending on the current state of the usb buffer, could lead to some time passing between bytes.
this might cause false positives on the esc timing out during the flash operation
https://github.com/BossHobby/QUICKSILVER/commit/074f1b54b83287e88649fc5f3150daad5cfb1377

i have now introduced some additional buffering which makes it so that the response is always sent in one go. if for nothing else, that seems to make the esc configurator much quicker to load the settings.