BossHobby/QUICKSILVER #77: Can't get into DFU mode

klavifon, 2023-01-23:
Hello,

yesterday I flashed a Quicksilver firmware on my Mobula 6. I was having trouble configuring it and decided to flash back to Betaflight. So today I decided I will try once more so I flashed it back to Quicksilver. Something went wrong during the flash procedure and now I can't put it in the DFU mode. I tried almost everything and I'm kinda hopeless to the point of getting myself a new FC.
So when I'm in the QS configurator I plug a USB into FC and reset it to bootloader. In ports I select a COM port which says QUICKSILVER, choose connect and it resets in Bootloader. Than I select the FW and the target and click Flash. And there is no COM port to select in this step. So I presume it didn't go in the DFU mode after all.
I'm using a Crazybee F4 (Flysky) with the ELRS EP2 RX on UART 1 so when my radio is turned on I get a solid light on the RX and a telemetry reading on my radio.

Please help because I'm becoming desperate.
Thanks

bkleiner, 2023-01-23:
I am sorry, but this not a support forum. Your fc not broken. Check our [docs](https://docs.bosshobby.com/Troubleshooting/) or hop on our [discord](https://discord.gg/8StVhvB6Tm).

closed 2023-01-23 by bkleiner

klavifon, 2023-01-23:
Thanks for the reply. I’m on macos unfortunately. I will try to fix this
issue on a Windows computer. Hope it works!
Best regards,
Ales