BossHobby/QUICKSILVER #84: OSD Fuel Gauge Volts Distinguish from Filtered Volts

brogrene, 2023-02-08:
OSD for Fuel Gauge volts can you put a gauge icon after the "V" so it's easily distinguishable from Filter volts when both are displayed?

bkleiner, 2024-01-22:
sorry, i forgot about this one. implemented as suggest in 
https://github.com/BossHobby/QUICKSILVER/commit/24a0ba60111f1550ff0208a4bc1ac00ae4e020ca
https://github.com/BossHobby/Configurator/commit/bfff6d7cb54efa58650a523bd2d2c8e93523b73d

closed 2024-01-22 by bkleiner

brogrene, 2024-01-24:
Awesome! Thanks for getting back to me. I look forward to release.
Is your current preference still on the GitHub or rather Discord for 
these requests? I only ask because Github looks to be >year since last 
"open" and I want to keep it simple for you and not duplicate if I have 
any other suggestions.

Regards,
Eric

from Filtered Volts (Issue #84)