BossHobby/QUICKSILVER #92: Add turtle mode active warning on OSD

Phjorttos, 2023-03-01:
Show OSD message when Turtle mode is activated via AUX switch to prevent rolling on the ground when forgot to switch off Turtle.

bkleiner, 2023-03-01:
i don't really want to permanently give up the screen real-estate, as even during turtle i think visibility is important (especially if you gotta do it quickly, eg during a race), but we could have the flightmode osd element display turtle while that is active.

Phjorttos, 2023-03-02:
I'm not flying whoops only bigger ones without proguards and I have turtle always on switch and like to have control over it when active or not (hanging in tree or safety net). Few times I forgot to turn off turtle when talking to someone or other hassle around after crash after gotten quad right side up. Got funny moments and scared other a bit when quad screams and rolls on ground. Gonna test more this behaviour. 
But maybe at least show it as flightmode could be ok. What I suggest: using turtle on AUX show warning as system message. Like it's showing warning when arm AUX on and throttle is raised.

closed 2023-12-09 by bkleiner