BossHobby/QUICKSILVER #95: Add stick commands for VTX control

pranas, 2023-03-22:
I'm using Quicksilver on an HDZero whoop, and I can't find an easy way to change the VTX channel. If I use the HDZero menu to change it, then Quicksilver still changes it back to whatever is set in its config. The only way I found is to hook up a computer and change it via Quicksilver configurator, but that's very inconvenient. Am I missing something, or it's the only way to do it?

bkleiner, 2023-03-22:
have you tried the Right-Right-Right gesture to enter the osd menu? https://docs.bosshobby.com/Features/#stick-gestures
first sub menu there will allow you to do the same thing as changing it via the configurator.

pranas, 2023-03-22:
This is exactly what I was looking for. Thanks, for your help.