BossHobby/QUICKSILVER #98: Integrated RX is not working on Cross F4 AIO

pranas, 2023-06-04:
I've flashed Quicksilver to Cross F4 board with an integrated serial ELRS receiver. I cannot get the RX to work: I've set the RX serial to USART_2, and I get RX_STATUS_DETECTING circling through protocols forever. I can see it going through CRSF and not picking it up. I tried Quicksilver 0.9.6 and 0.9.6-dev. I tried with the ELRS version it shipped with (3.0.1), and I tried updating to 3.2.1, but nothing changed. The RX is fine, as I was able to set the binding phrase via WiFi, and I got it connected to my controller. Any other ideas to try?

bkleiner, 2023-06-05:
I am sorry, but this is not a support forum.
Consider hopping on our [discord](https://discord.gg/8StVhvB6Tm), usually there is somebody around to help you out.

PS: check if you have "model match" enabled, the correct target flashed (crossf4) and did select an appropriate uart (iirc uart2) on the setup page.

closed 2023-06-05 by bkleiner

pranas, 2023-06-05:
I suspected it could be a bug in Quicksilver. Sorry if you get a lot of invalid reports like this.

Your hint to check "model match" was a good one. Indeed Happymodel shipped the board with "model match" enabled, which was not overwritten with the firmware update.

[image]