BossHobby/QUICKSILVER #112: blackbox log download fail

amitleibov, 2024-01-14:
I record blackbox logs, they appear to be on the FC, however downloading them ends with an error message "blackbox download failed"
what could be the reason ?
FC JHEMCUF405PRO target  quicksilver ver 0.10.2
[image]

amitleibov, 2024-01-14:
[image]

amitleibov, 2024-01-14:
[image]

amitleibov, 2024-01-14:
I 
[image]
can confirm that on another drone it works fine .... I'd still want to know if there is a bug that can be fixed, as I want blackbox to analyze my drone build.. Thanks.

bkleiner, 2024-01-14:
There are a number of reasons why the download failed, however most common seems to be that some flight-controller/computer combinations do not fully deliver the data to the browser. 

Could you:
a) try to download the existing log a couple more times, aforementioned problem does not manifest consistently.
b) try recording another (shorter) log and try downloading then.

amitleibov, 2024-01-17:
thanks for the fast response. deleted the files and tried another flight and this time downloaded as expected.

closed 2024-01-17 by amitleibov