BossHobby/QUICKSILVER #118: [help] platformIO build error

vangao-gg, 2024-10-25:
hi everyone,
QS is a very good project, and I want to try compiling the latest code in Windows, but encountered the following error message during compilation:
[image]

1. The commit version is: bump version; c5d58c21cb156c4caa15ab2a72b12445915ede15
2. platformIO core 6.1.16 and home 3.4.4
3. The list results are as follows:
[image]

4. I switched the branch of the repository to 0.10.0, and the compilation is normal

bkleiner, 2024-10-25:
interesting. i dont think the commit you point out is at fault - that file indicates the version to the configurator and is of no relevance for pio. 
can you try deleting the `.platformio` folder in your windows user folder? looks to me maybe something got corrupted during download.

vangao-gg, 2024-10-25:
Thank you, bkleiner. Following your instructions, I deleted the. platformio fold and reconfigured it. The build is now normal. Thank you again and have a nice day;

closed 2024-10-25 by bkleiner